iris-ua/iris_lama_ros #1: Allow compilation in Kinetic

facontidavide, 2019-10-08:
Hi,

I am trying to compile in Kinetic. A couple of changes to make it easier for everybody:

    lower minimum CMAKE requirement to 3.5 (including repository **iris_lama**)
    add set(CMAKE_CXX_STANDARD 11) to CMakeLists.txt iris_lama_ros.

eupedrosa, 2019-10-08:
Are those the only requirements to make it work with Kinetic? Right now I do not have a Kinetic machine to test it.

I've tested your changes with Melodic and it works fine. If you can confirm that it works with Kinetic I will push the changes.

facontidavide, 2019-10-08:
I am also having problem finding Eigen3, but that is perhaps my fault.

You can easily allow continuous integration with Travis CI using the ros-industrial repository

https://github.com/ros-industrial/industrial_ci

In this way you can continuously check the compatibility with multiple ROS distros.

sweetquiet, 2019-10-09:
@facontidavide hello I am also having problem finding Eigen3,What should I do? thank you!

closed 2019-10-22 by eupedrosa