iris-ua/iris_lama_ros #3: Question: how hard would it be to add 3D mapping and localization?

facontidavide, 2019-11-14:
it is very nice to see haw there is a clear separation between the ROS and non ROS part in LAMA.

My question is: how mature and tested is the 3D (octomap) implementation of iris_lama? Do you see a transition to 3D SLAM with sensors such as Velodyne something very hard or relatively straightforward?

Just want your opinion before entering the white rabbit's hole myself.

eupedrosa, 2019-11-14:
Hi,

> how mature and tested is the 3D (octomap) implementation of iris_lama?

The best I can say about this is that, once I got it working, It never failed me. But this is not enough to say it is mature or was properly tested. For 3D applications I did mapping with known poses and this is the result:
[image]

[image]

[image]

Octomap's authors made this datasets available.
My solution/implementation for 3D mapping is more flexible, faster and uses less memory than octomap but it lacks the testing and maturity of OctoMap.

> Do you see a transition to 3D SLAM with sensors such as Velodyne something very hard or relatively straightforward?

You have reached my personal white rabbit's hole. But I am yet to completely enter the hole.

Using data from a Velodyne should be straightforward, even for 2D I use a point cloud to represent the data. Unfortunately I do not have a Velodyne, nor a variety of dataset, for testing. 

Expanding to 3D SLAM is something I am genuinely interested in doing. Is this something you are interested in?

facontidavide, 2019-11-14:
> I did mapping with known poses 

I am not sure what you mean by "known poses".

> Unfortunately I do not have a Velodyne, nor a variety of dataset, for testing.

I can give you my rosbags :)

> Expanding to 3D SLAM is something I am genuinely interested in doing. Is this something you are interested in?

I have a use case in which the robot is an autonomous vehicle and i want to have a very good localization (as precise as possible).
2D data is not rich in terms of features, but 3D data seems to be very rich in terms of natural landmarks that would unambiguously localized the robot.
I worked a lot to improve Lego-LOAM: https://github.com/facontidavide/LeGO-LOAM-BOR

But I am too frustrated to go on: the original code is too complex, too many hard-coded parameters and I can not save maps and have a localization-only mode.

I would rather use my time to build on top of iris_lama, rather than continue refactoring Lego-LOAM.
I think if would be better for my use case and for the community.

But this might be based on a positive bias I have toward your software ;)

eupedrosa, 2019-11-14:
> I am not sure what you mean by "known poses".

Fancy works to say that I have a ground truth.

> I can give you my rosbags :)

I would appreciate that. Can you provide them to me when possible?

> I would rather use my time to build on top of iris_lama, rather than continue refactoring Lego-LOAM.
I think if would be better for my use case and for the community.

You have my attention. It has been my goal to extend it to 3D. This could be the final push to finally dive in.

Note that LaMa has saving and loading implemented but it is not active in the code. It lacks a clear structure.

salsicha, 2020-02-28:
+1 for a 3D implementation. I like the choices you made, particularly SDM. I work with a combination of Cartographer and VoxBlox and would love to see a single project that is a compromise between the two.

Is there any update on making 3D work?

eupedrosa, 2020-02-28:
Hi @salsicha.
Sorry but no *visible* update on 3D. It is on my TODO list and I hope to start working on it soon. 
Right now I'm finishing another project and it is not yet a priority. 

> I work with a combination of Cartographer and VoxBlox

Nice, I would like to take a look at it, is it public?

PS: **salsicha** is a funny name in my language https://pt.wikipedia.org/wiki/Salsicha

salsicha, 2020-02-29:
> > I work with a combination of Cartographer and VoxBlox
> 
> Nice, I would like to take a look at it, is it public?

Sorry, no code. But here’s a picture:
https://newscenter.lbl.gov/wp-content/uploads/sites/2/2019/10/PRISM-ErikaSuzuki_1000px.jpeg

> PS: **salsicha** is a funny name in my language https://pt.wikipedia.org/wiki/Salsicha

Its a nickname, it’s a reference to this:
https://pt.m.wikipedia.org/wiki/Scooby-Doo