iris-ua/iris_lama_ros #5: Question: position updates from other system

JaccovdS, 2019-11-27:
Is it possible to update the localization with position updates from other sensors or systems (e.g. GPS)? 
Something similar can be found in for example Cartographer: https://google-cartographer-ros.readthedocs.io/en/latest/going_further.html#more-input

eupedrosa, 2019-11-27:
Right now, it is _not_ possible to update the localization with other sources of information.

JaccovdS, 2019-11-27:
Would it theoretically be possible to adjust the system such that it can take advantage of other sources of information or would this require a complete redesign of the algorithms?

facontidavide, 2019-11-27:
my 2 cents:

1) Graph based SLAM is usually more suitable for the integration of heterogeneous sensors.
2) In you specific case, the low-hanging fruit (that DOES NOT mean "quick and dirty") would be to use **robot_localization** to create an odometry that takes advantage of the GPS.

http://docs.ros.org/melodic/api/robot_localization/html/integrating_gps.html

eupedrosa, 2019-11-27:
@facontidavide *stole* my answer with number 2:
> 2. In you specific case, the low-hanging fruit (that DOES NOT mean "quick and dirty") would be to use robot_localization to create an odometry that takes advantage of the GPS.

facontidavide, 2019-11-27:
["Nerd people think alike"](https://www.phrases.org.uk/meanings/great-minds-think-alike.html)

facontidavide, 2019-11-27:
@eupedrosa , actually it would be nice to add a section in the README.md with Frequently Asked Questions and mention this.

reinzor, 2019-11-28:
Thanks for the response. We are aware of the EKF package robot localization. However, we are looking for a localization method that can process lidar measurements (using scan matching on an occupancy grid) and absolute pose measurements (coming from a beaconing system for example). 

IMO, for the localization node `loc2d_ros` it makes sense to fuse this information if the occupancy grid and the absolute measurement information is aligned. We could create a measurement model that scores that particle based on the (most probably) gaussian distribution  of the incoming absolute measurement. However, I do not know how this would work for the `slam` case. 

What do you think?

eupedrosa, 2019-11-28:
Hi @reinzor.
In this SLAM solution, localization and mapping are decoupled. You do localization then mapping. In theory, you can use whatever localization solution you want. So, if you change `local2d_ros` to accommodate GPS info you can use the same algorithm in `slam`. 

@facontidavide :
> actually it would be nice to add a section in the README.md with Frequently Asked Questions and mention this.

:+1: As soon as I have some time, I will add it to the README.

reinzor, 2019-11-28:
Thanks for your response. I see. Would you accept a pull request that integrates this functionality?

- Add another update method for [PoseWithCovarianceStamped](http://docs.ros.org/melodic/api/geometry_msgs/html/msg/PoseWithCovarianceStamped.html) or [Odom](http://docs.ros.org/melodic/api/nav_msgs/html/msg/Odometry.html) in `loc` and `slam` (same as https://github.com/iris-ua/iris_lama/blob/master/src/loc2d.cpp#L92 and https://github.com/iris-ua/iris_lama/blob/master/src/slam2d.cpp#L87)
- Add an `onOdom` / `onPoseWithCovariance` callback method (same as https://github.com/iris-ua/iris_lama_ros/blob/master/src/slam2d_ros.cpp#L111 , https://github.com/iris-ua/iris_lama_ros/blob/master/src/loc2d_ros.cpp#L103) that calls the internal library

facontidavide, 2019-11-28:
> Add another update method for PoseWithCovarianceStamped or Odom in loc and slam

@reinzor do not forget that iris_lama is (and must remain) ROS independent. We should not add ROS specific messages to that API.

> However, we are looking for a localization method that can process lidar measurements (using scan matching on an occupancy grid) and absolute pose measurements (coming from a beaconing system for example).

This is conceptually a contradiction in terms. In **localization** (not mapping) which information should you trust more, scan matching or absolute position?

reinzor, 2019-11-28:
> @reinzor do not forget that iris_lama is (and must remain) ROS independent. We should not add ROS specific messages to that API.

Sure

> This is conceptually a contradiction in terms. In localization (not mapping) which information should you trust more, scan matching or absolute position?

If the measurement uncertainties / measurement models are correct. The filter outcome should represent the best possible estimate in my opinion.

facontidavide, 2019-11-28:
> If the measurement uncertainties / measurement models are correct. The filter outcome should represent the best possible estimate in my opinion.

You just described what an EKF is supposed to do  ;)

reinzor, 2019-11-28:
Just to clarify, what is the conceptual difference between a particle filter and kalman filter. They both try to robot state but with different measurement models / measurements. I don't see a conceptual difference. What would you suggest? Adding a KF on top of this filter?

eupedrosa, 2019-11-28:
> If the measurement uncertainties / measurement models are correct. The filter outcome should represent the best possible estimate in my opinion.

That is the theory. But remember that we are using the expected value of that measurement and not the distribution. 

> Just to clarify, what is the conceptual difference between a particle filter and kalman filter. They both try to robot state but with different measurement models / measurements

KF is uni-modal and PF is multi-modal.

reinzor, 2019-11-28:
> That is the theory. But remember that we are using the expected value of that measurement and not the distribution.

Doesn't this happen in a KF aswell, by calculating the "innovation" in the measurement model. The measurement models serves the expected value, only the scoring is different since we are using a different state representation. 

> Yes, I am willing to accept.

Thanks. We are looking for a solution where we can leverage global pose information when there are no obstacles present in the scene / map. IMO, this can only be obtained by fusing this information directly in the particle filter.

reinzor, 2019-11-28:
A second option I think of now is using a KF for fusing information and then feed this estimate to this particle filter but it feels not really optimal. What are your thoughts?

eupedrosa, 2019-11-28:
Well... In my opinion it is not optimal. But you can prove me wrong.. Employ your ideas, have some results, than we can discuss the outcome.

closed 2021-11-19 by eupedrosa