iris-ua/iris_lama_ros #7: How can iris_lama be automatically linked with ROS2 'colcon' tool?

bluemoon93, 2020-03-26:
How can the `iris_lama` package be automatically linked, when using `colcon` to build the ROS2 port of `iris_lama_ros`? 

My current approach is to copy header files into the `include/lama` folder, and building `iris_lama.a` to put into the src directory. This works, but is not very portable. Is there a way to automate this, and build both packages automatically?

eupedrosa, 2020-03-26:
What is the output of the error?

bluemoon93, 2020-04-17:
There isn't an error, I fixed all linkage errors by building `iris_lama` separately with `cmake` and 
  
    cp iris_lama/build/src/libiris_lama.a dev_ws/src/iris_lama_ros
    cp -r iris_lama/include dev_ws/src/iris_lama_ros/

If I don't copy the header `.h` files manually, when compiling I get

    fatal error: lama/slam2d.h: No such file or directory

And if I don't copy the binary `libiris_lama.a` manually, I get 

    No rule to make target '/mnt/myfolder/dev_ws/src/iris_lama_ros/libiris_lama.a', needed by 'src/loc2d_ros'

I think there are ways of automating this and, when building `iris_lama` with catkin, making it available to colcon? If not, the documentation includes all these steps. It's just not as elegant,

eupedrosa, 2020-04-17:
Yes, it is not elegant. I will try to find a fix.

eupedrosa, 2020-04-17:
@bluemoon93, I think I solved it. I made some changes to the packages to make it work. I am going to push the changes. With this push it will no longer be necessary to do the manual copies. Nonetheless, can you also verify that it is working? Remember to pull the changes from `iris_lama` before testing.

CleopatraDion, 2020-04-18:
Just as some input. I pulled the eloquent-devel branch to build it again, following the instructions and I was getting some linkage errors. Then I found this issue and built iris_lama with colcon, and then iris_lama_ros, as per your new method and it was fine. So, instructions will prob have to be updated. I will be happy to help out with updating instructions and things like that once I get a bit more used to the package.

eupedrosa, 2020-04-18:
Hi, @CleopatraDion.

> I will be happy to help out with updating instructions and things like that once I get a bit more used to the package.

That would be great. I did not update the instructions because I was waiting for feedback. If it works fine and you are willing to do the necessary changes to the documentation, just do a pull request and I will accept it.

bluemoon93, 2020-04-20:
Confirmed this worked on both branches. I've updated docs and am closing the issue

closed 2020-04-20 by bluemoon93