iris-ua/iris_lama_ros #8: ros2 launch

CleopatraDion, 2020-04-17:
Hi,
I'm working with this on the eloquent devel. @bluemoon93 I see that on the dashing-devel you added 2 launch files for live slam. When I build this on eloquent it didn't pick up these launch files because **_launch_** wasn't picked up in the name (pf_slam2d_live.py). I changed this to pf_slam2d_live_launch.py and everything was fine. Did you not have this issue building on dashing?

bluemoon93, 2020-04-17:
I've added the files now, I add forgotten to add them (because I was testing things on dashing-devel). If you pull, everything should be there. They don't need to have **launch** in their names, as far as I know, and you can run things with `ros2 launch iris_lama_ros2 slam2d_live.py`. 

Let me know if things are correct now and I will close this issue.

CleopatraDion, 2020-04-17:
I tested it there again, with no _**launch**_ included in the name it wouldn't add them to the install launch director. When i added it, it then appears. I am using ubuntu 18.06 with eloquent. 
I also deleted and cloned the folder again and I got the same issue... no launch in the name and it doesn't get put into the launch folder... strange, but this is maybe not a question for here

bluemoon93, 2020-04-17:
I wasn't able to replicate this. I used the attached Dockerfile with Ubuntu 18.06 and ROS2 Eloquent, and this worked

    # colcon build
    # . install/setup.bash
    # ls install/iris_lama_ros2/share/iris_lama_ros2/launch/
    loc2d.py  pf_slam2d_live.py  pf_slam2d_offline_launch.py  slam2d_live.py  slam2d_offline_launch.py

That being said, it's an easy fix, if all that's needed is to rename the files. I've pushed the commit and am closing this issue now.

closed 2020-04-17 by bluemoon93