iris-ua/iris_lama_ros #10: Internal comiler error when building for eloquent

RodBelaFarin, 2020-05-18:
I get the following error output for building iris_lama_ros2 with colcon build:

`In file included from /opt/ros/eloquent/include/rclcpp/subscription_base.hpp:27:0,
                 from /opt/ros/eloquent/include/rclcpp/callback_group.hpp:26,
                 from /opt/ros/eloquent/include/rclcpp/any_executable.hpp:20,
                 from /opt/ros/eloquent/include/rclcpp/memory_strategy.hpp:24,
                 from /opt/ros/eloquent/include/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/eloquent/include/rclcpp/executor.hpp:33,
                 from /opt/ros/eloquent/include/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /opt/ros/eloquent/include/rclcpp/executors.hpp:21,
                 from /opt/ros/eloquent/include/rclcpp/rclcpp.hpp:145,
                 from /home/mascor/turtlebot3_ws/src/iris_lama_ros/include/lama/ros/slam2d_ros.h:37,
                 from /home/mascor/turtlebot3_ws/src/iris_lama_ros/src/slam2d_ros.cpp:34:
/opt/ros/eloquent/include/rclcpp/experimental/intra_process_manager.hpp: In member function ‘void rclcpp::experimental::IntraProcessManager::do_intra_process_publish(uint64_t, std::unique_ptr<MessageT, MessageDeleter>, std::shared_ptr<typename std::allocator_traits<_Alloc>::rebind_traits<MessageT>::allocator_type>) [with MessageT = nav_msgs::msg::OccupancyGrid_<std::allocator<void> >; Alloc = std::allocator<void>; Deleter = std::default_delete<nav_msgs::msg::OccupancyGrid_<std::allocator<void> > >]’:
/opt/ros/eloquent/include/rclcpp/experimental/intra_process_manager.hpp:237:3: internal compiler error: Segmentation fault
   }
   ^
`

eupedrosa, 2020-05-18:
Hi @RodBelaFarin.

Can you provide a little bit more of information? Such as the OS version and ros version (from the logs I assume eloquent). 

I did a fresh compilation of the code and I was not able to reproduce the problem.  If this is a compiler problem may not be able to help. Furthermore, the problem seems to originate inside the ROS client library, meaning that is has nothing to do with with this package.

RodBelaFarin, 2020-05-18:
Yes - eloquent and Ubuntu 18.04.

and pretty weird. i am able to compile other packages as usual. do i need to build the lama library additionally with cmake? or is it included in the build with colcon, if i just clone 2 repos in my workspace?

eupedrosa, 2020-05-18:
Just clone the 2 repos to your workspace. It works with colcon. For `iris_lama_ros` you are using the `eloquent-devel` branch, correct?

This may be a stupid question, but, do you have enough memory in your system while building?

RodBelaFarin, 2020-05-18:
No, that was actually a very good question! I am running the system in a VM and there seemed to be some issue with the space. I rebooted the VM and now it is compiling fine, thanks!

closed 2020-05-18 by RodBelaFarin