iris-ua/iris_lama_ros #15: Produced map doesnt match with the Laserscan

pfedom, 2020-07-29:
Hi, 
I just started using lama. I am trying to use the particle filter SLAM on Ros2 eloquent (Ubuntu 18.04). 
My Laser is mounted upside down on the robot and there is a static transform that is taking care of this. 
But Lama is producing a map, that is flipped back around and I have no idea where this is coming from. 
I attach a picture, where you can see my laserscan and the map that is produced shown in rviz. It doesnt match. 
The static tf should be correct. I proved it with slam_toolbox and it worked fine. 
The static tf that is flipping 180° is from base_scan -> base_link. 

Any idea how I could fix this? 
Thanks in advance! 

[image]

eupedrosa, 2020-07-29:
Hi @pfedom,

I am aware of this problem. This was fixed with 594b91957ff12dcb0c3bf46fbee7023865da9c12. But this is yet to be merged with the **ROS2** port. 

I will try to fix this as soon as possible.

eupedrosa, 2020-07-29:
Hi @pfedom.
Commit a1d7f699f61da9a14045a9adbf1c83a30465bba3 fixes the problem (I hope). Can you please update you code and verify this works for you?

pfedom, 2020-07-30:
It's not working. Still the same bug. 
But I see you only updated the slam2d files. I am using the particle filter slam (pf_slam2d). Or do these files take effect on the particle slam too? 

The "normal" slam2d isnt working for me. The map frame is not being published. The pf_slam2d works (except for the flipping error). 
Both spam this info, but I think thats not an issue: 
[INFO] [iris_lama_ros2.pf_slam2d_ros]: Message Filter dropping message: frame 'base_scan' at time 1596102489.100 for reason 'Unknown'  

Should I put up another issue for this problem or do we discuss this here too?

eupedrosa, 2020-07-30:
I forgot to change in pf_slam2d. Please try the update.

> The "normal" slam2d isnt working for me. The map frame is not being published. The pf_slam2d works (except for the flipping error).

It was supposed to work if it work for pf_slam2d.

pfedom, 2020-07-30:
pf_slam2d works now as expected. 
Thank you!

closed 2020-07-30 by pfedom