iris-ua/iris_lama_ros #16: Using another laserscan topic than /scan

pfedom, 2020-07-29:
Hi, 

I am trying to get lama workin with another laserscan topic than /scan. 
Where do I tell the SLAM which topic is providing the laserscan data?
I am using Ros2 Eloquent on Ubuntu 18.04. 

I tried to include `scan_topic: scan_new` and also  `scan_topic: /scan_new` in the config file but if I do this the slam doesnt produce anything (not even a map frame) without throwing errors in the terminal. 

Thanks in advance!

eupedrosa, 2020-07-29:
Hi @pfedom.

> I am using Ros2 Eloquent on Ubuntu 18.04.

The ROS 2 version is not yet fully supported. It was a *proof of concept* port that we did to verify how *easy* it could be done.
But I think you can always use [topic remapping](https://index.ros.org/doc/ros2/Tutorials/Node-arguments/#name-remapping).

pfedom, 2020-07-30:
okay thanks!

closed 2020-07-30 by pfedom