iris-ua/iris_lama_ros #18: [signalFailure] Drop message: frame 'base_scan' at time 9.005 for reason(0)

atinfinity, 2020-09-19:
I tried to use slam2d_ros(dashing-devel) with TurtleBot3(Gazebo).  
https://github.com/atinfinity/turtlebot3_lama_ros2

But, the following message was displayed.

```
[slam2d_ros-1] [INFO] [iris_lama_ros2.slam2d_ros]: [signalFailure] Drop message: frame 'base_scan' at time 9.005 for reason(0)
[slam2d_ros-1] [INFO] [iris_lama_ros2.slam2d_ros]: [signalFailure] Drop message: frame 'base_scan' at time 9.206 for reason(0)
```

On the other hands, I changed to pf_slam2d. It works fine.

takahashi-e6, 2021-02-08:
I changed the file as follows and it worked. Maybe the timeout time was too short.

iris_lama_ros/src/slam2d_ros.cpp(line 111)
```
tf_buffer_ = std::make_shared<tf2_ros::Buffer>(node->get_clock(), tf2::Duration(30));
```
↓
```
 tf_buffer_ = std::make_shared<tf2_ros::Buffer>(node->get_clock());
```

atinfinity, 2021-02-09:
@takahashi-e6 Thank yor for your comment. I will check again.