iris-ua/iris_lama_ros #25: Fixed Frame [map] does not exist - is LaMa localization node supposed to publish /map --> /odom ?

WilliamWoo45, 2020-11-24:
Hi! 
Firstly, thanks for providing such great SLAM / Localization work.

I was trying to implement the `localization function` of LaMa ROS based on my own occupancy grid map. I launched the command lines in different windows in the following order:
1. `rosrun iris_lama_ros loc2d_ros scan:=base_scan`
2. `rosrun map_server map_server /opt/ros/kinetic/share/husky_navigation/maps/mymap.yaml`
3. `rviz `
4. `rosbag play --clock /home/williamwoo/Z_husky_S2.1/S2.1_new/husky2-pose-_2020-06-12-s2.1-way11-mrpt.bag`

The map is successfully loaded, where in the command line saying that: 
```
[ INFO] [1606198756.368682224]: Requesting the map...
[ INFO] [1606198761.232647620, 1592210365.736253594]: Localization parameters: d_thresh: 0.01, a_thresh: 0.20, l2_max: 0.50
[ INFO] [1606198761.454955949, 1592210365.938106792]: 2D Localization node up and running
```

I suppose the localization node is successfully running. However, there is an error saying that `Fixed Frame [map] does not exist` (as shown below). 
[image]

The tf is:
[image]

The rosbag info is: 
[image]

So I'm wondering how could I solve this problem? I suppose what the LaMa ROS localization node does is to publish /map --> /odom, right?
The rosbag that I use is [here](https://github.com/WilliamWoo45/iris_lama_ros/blob/master/mymap.yaml) and the map that I used is [here](https://github.com/WilliamWoo45/iris_lama_ros/blob/master/husky2-pose-_2020-06-15-S1-way11-test.bag).

Thanks and all the best!

Best Regards,
William

eupedrosa, 2020-11-24:
Hi,
Thank you for using LaMa.

I think your problem is here: `rosrun iris_lama_ros loc2d_ros scan:=base_scan`.
Your bag has a laser scan in the `/scan` topic, so you should use that topic. The `/map` frame does not exist because the localization never received a laser scan.

Please try running with `rosrun iris_lama_ros loc2d_ros scan:=/scan` and let me know if it works.

WilliamWoo45, 2020-11-24:
Hi Eurico,

Thanks a lot for your speedy response. 
It works! I did not expect the problem could come from there :)

BTW, I did the global localization tests and found that, without specifying the correct initial pose, the `base_link` and `odom` frames would **jump to the correct place** for 1-2 seconds at the beginning of some of my rosbags. Then they would jump back to somewhere near the origin (0,0), and the localization failed (see examples below).
Example 1:
[image]

[image]

Example 2:
[image]

[image]

But at some starting place near the origin of the map (0,0), LaMa Localization managed to do the global localization after the robot moving for around 3-5m (see the screenshots below):
[image]

So, I'm wondering does the current `localization node` actually support **Global Localization** (i.e. localization without specifying initial pose) to some extent?

Cheers!

Best Regards,
William

eupedrosa, 2020-11-24:
The current `localization node` does _not_ have support for **Global Localization**. So you must provide an initial pose, or you can use rviz for that.

However, there is a global localization mechanism already implement in the base code of the localization algorithm. It does not requires an initial pose. It may find the correct pose, but is is not guarenteed. There is a hard-coded threshold that may not be suitable for all applications. 

I may implement the global localization in the ROS node. It is not a dificult task. Stay tuned.

WilliamWoo45, 2020-11-24:
Hi Eurico,

Noted with thanks.
I'll stay tuned and I'm very looking forward to your updates on global localization in the ROS node, since it is quite necessary when the robot is lost track of or when the robot is kidnapped in the practical applications.

Thanks again and take care!

Best Regards,
William

eupedrosa, 2020-12-05:
@WilliamWoo45, you can now try a global localization method for the localization node 242c3fca21da5bf4a6cbb089b0e8f41d03767004.

Please notice that the `~gloc_thresh` parameter may need to be adjusted to your application.

WilliamWoo45, 2020-12-05:
Hi Eurico,

Thanks a lot for your speedy coding.
Noted. I'll try it as soon as possible.

Best Regards,
William

closed 2020-12-16 by eupedrosa

WilliamWoo45, 2020-12-17:
Hi Eurico @eupedrosa ,

I've implemented the `global localization` approach. Its localization success rate is around 10% ~ 15% in the **repetitive** corridor environment (as shown previously) **without** specifying the `initial pose`. I believe it can achieve **much better** `global localization` performance in **ordinary** environments with sufficient distinct geometric features, and with better adjustment/tunning of the `~gloc_thresh` parameter.

Regarding the `global localization` ability, I think `LaMa ROS` performs pretty good and is around the same performance level with `AMCL` or `MRPT_localization`. 

Thanks and all the best!

Best Regards,
William