iris-ua/iris_lama_ros #30: Failure on updating map with loc2d_ros

lantica, 2021-02-26:
I am loofing at loc2d_ros for replacing the amcl node.
It works quite well even with the default params.
However, the localisation will fail if I publish a new map to the map topic.
Any idea?

The ros version using is Noetic.
The params using are as below:
```
                    global_frame_id: "map",
                    odom_frame_id: "odom",
                    base_frame_id: "base_link",
                    scan_topic: scanTopic,
                    d_thresh: 0.1,
                    a_thresh: 0.2,
                    l2_max: 0.5,
                    strategy: "gn",
                    use_map_topic: true,
                    first_map_only: true,
                    use_pose_on_new_map: false,
                    force_update_on_initial_pose: false,
```

I can send you the rosbag file if needed.

eupedrosa, 2021-02-26:
Can you share the rosbag? I will take a look into it.

lantica, 2021-03-02:
I found I made a stupid mistake.
It works after I set ```first_map_only``` to false.
I will close the issue.

closed 2021-03-02 by lantica

eupedrosa, 2021-03-02:
It is not a *stupid* mistake, it is a *learning* mistake.