iris-ua/iris_lama_ros #36: covariance of position

TZVIh, 2021-04-06:
Hi,
I am working with this package.
How can I know the covariance of the solution?
Thanks

TZVIh, 2021-04-06:
Hi,
Thanks for the fast response!
I am using the localization mode.

I checked the "/pose" topic and the covariance is always zero.
attached pic of echo to this topic
[image]

eupedrosa, 2021-04-06:
Are you using the master branch? The feature is very recent.

TZVIh, 2021-04-06:
I just pull the last commit.
The build failed (catkin build or catkin make)
[image]

eupedrosa, 2021-04-06:
You also have to use the latest version from https://github.com/iris-ua/iris_lama

TZVIh, 2021-04-06:
works!!!!!!
Thanks!

TZVIh, 2021-04-07:
one more question, 
why the frame of the pose msg is empty?

eupedrosa, 2021-04-07:
It should not be empty, it should have the global frame id. It is a bug, I will fix that.

TZVIh, 2021-04-11:
Hi,
I want to share with you that I tested LaMa's localization mode against AMCL and other packages and I must say that LaMa's performance was the best.

I'm checking now the covariance of the solution.

I set the initial pose to 0.0 (this is not the right pose) on the launch file (loc2d) 

At the beginning of the bag file, the variance of the variables (diagonal) is 1.
[image]

When the robot starts to move the variance drops to a very small value, you can see that the robot didn't localize right.
[image]
[image]

If I understand correctly I would expect to see that the variance is still high because the robot did not find its correct position.

Can you explain to me why this is happening?

Thanks!

TZVIh, 2021-04-13:
Hi @eupedrosa,
Can you please help me?

eupedrosa, 2021-04-13:
Hi @TZVIh, yes I can help you.

The covariance that I publish results from the Information matrix used in the optimization. As such, it will reflect the end state of the optimization. The optimization will always find a local minimum, therefore it will most likely underestimate the true covariance.

But I believe there are 2 changes that you can try to see if the covariance raises when it is on a bad location:

1. Increse the value of the parameter `~l2_max` to `5.0` or even `10.0`. This will increase the time it takes to load the map, but it is OK because it happens only once. Whan happens is that for a small value such as `0.5` (the default), on a bad localization situation most laser points will have null gradient, and the gradient contributes to the covariance. By increasing this parameter you will be adding more points to the calculation of the covariance.
2. The covariance is scaled for robust fitting. Maybe this should not be done when return the covariance. Go to  
https://github.com/iris-ua/iris_lama/blob/37f49b91c70aafbc91a6a713dac09037026ea146/src/nlls/solver.cpp#L106 and remove lines 108, 109 and 109.

TZVIh, 2021-04-13:
Hi,
Thanks for the response.

I changed l2_max to 100.0. in addition, I removed lines 108-110 from the solver.cpp as you write.

The variance still has a small value (0.002)

Do you have more ideas?

eupedrosa, 2021-04-13:
Changin `l2_max` to 100 is not a good idea, it will create a huge distance map. 10 is a better number.

Like I said, the optimization covariance will underestimate the true covariance. It will not provide the measurement that you are expecting. The only way to have something similar to what you are expecting, is to sample around the calculated pose and estimate a covariance weighted by the likelihood.

TZVIh, 2021-04-13:
I really appreciate your help. 
I will do it.

closed 2021-04-13 by TZVIh