iris-ua/iris_lama_ros #37: Optionally disable map->odom tf transfrom

Thazz, 2021-05-06:
I would propose to add a parameter to optionally disable publishing of `map->odom` tf transfrom, such as `tf_broadcast` parameter  in [amcl](http://wiki.ros.org/amcl) package.

Use case for this is when one would be using [robot_localization](http://wiki.ros.org/robot_localization) package for fusing multiple localization sources with Kalman filter (e.g localization with lama + bluetooth beacons or GPS). In this case `map->odom` is (or can be) published by `robot_localization` nodes and should be disabled for `robot_localization` input sources, i.e lama ros.

eupedrosa, 2021-05-06:
Hi @Thazz 

Are you talking about a specific node, for example `loc2d_ros`, or should this feature be available in all nodes?

Thazz, 2021-05-12:
I think it only makes sense to implement it in `loc2d_ros` as it is used for localization. 

For `slam2d_ros` it could be useful  in case od lifelong mapping as discussed in https://github.com/iris-ua/iris_lama_ros/issues/38, where this node would also be used for localization, but as I understand from discussion, LaMa is currently not suitable for this.

Another question, is it possible to estimate covariance matrix for localization, so that pose could be published as `geometry_msgs::PoseWithCovarianceStamped` message?

eupedrosa, 2021-05-13:
I think I can add the option to disable tf broadcasting for `loc2d_ros`.

> Another question, is it possible to estimate covariance matrix for localization, so that pose could be published as geometry_msgs::PoseWithCovarianceStamped message?

This already exist for `loc2d_ros`

commit 3b4325fa (2021-11-19): [loc2d] Add option to disable map->odom tf (closes #37).
closed 2021-11-19 by eupedrosa