iris-ua/iris_lama_ros #38: LifeLong Mapping

TZVIh, 2021-05-09:
Hi,
If I built a map and I work with localization mode.
The area has changed and I am interested in updating the map while driving.
Is there such a possibility?

facontidavide, 2021-05-09:
I have been toying with keeping the mapping always on.
It was working surprisingly well, but my two cents are that you never really know if you map gets corrapted, without a "human in the loop".
To be fair, I expect any SLAM that claims to have "lifelong mapping" to face similar problems.

@eupedrosa may have a different opinion, but I think that lifelong mapping mkes more sense in a SLAM implementation that uses graph slam under the hood.

eupedrosa, 2021-05-09:
I think that for "lifelong mapping" you need some mechanism where areas of the map can be "forgotten" so it can change over time. 

LaMa SLAM assumes that the environment is static with some "sprinkles" of dynamic objects, e.g. a person walking by.
Furthermore, the more it senses an area the more difficult it is to change it. This happens because it uses an occupancy map that is based on how frequently a cell is marked as free or occupied, all without an upper bound.

If the environment quickly changes its geometry, it may work. But I would not bet on it.

And just like @facontidavide said, there is always the possibility to have a corruped map. If that happens, it may never recover.

Graph SLAM may (indeed) be the best approach. But LaMa does not uses graphs (maybe in the future ????).

TZVIh, 2021-05-09:
@facontidavide  @eupedrosa Thanks for your answers!
I have currently built a map of several rooms and I am interested in adding the rest of the rooms on the floor to the map.

From the answers, I understand that it is better for me to build a new map or work with graphs.

I'm really happy with LAMA's performance so I do not want to move to another approach :)