iris-ua/iris_lama_ros #40: No map publish when use with real hardware

IST-DEV20, 2021-06-15:
How to fix this no map publish when use with real hardware? (can be used with gazebo)

jcmonteiro, 2021-06-15:
@IST-DEV20, this is likely a problem in your setup. Check if the lidars are published to the correct topic and with data and also if you have your `base_link` to `odom` tf being updated as you move the robot.

IST-DEV20, 2021-06-15:
@jcmonteiro, topic and data is look good but when I launch /slam2d_ros I can't echo tf between `base_link` to `odom`
my tf tree
[image]
my /scan topic
[image]
[image]

jcmonteiro, 2021-06-15:
Could you share your roslaunch configuration and the rosnode graph?

IST-DEV20, 2021-06-15:
@jcmonteiro this is my launch of `slam2d_ros.launch`
`
<?xml version="1.0"?>
<launch>
    <arg name="scan_topic" default="/scan"/>
    <arg name="rosbag" />

    <arg name="particles" default="30"/>
    <arg name="d_thresh"  default="0.5"/>
    <arg name="a_thresh"  default="0.5"/>
    <arg name="mrange"    default="20"/>

    <arg name="threads"   default="4"/>
    <arg name="use_compression" default="false"/>

    <param name="/use_sim_time" value="true"/>

    <node pkg="iris_lama_ros" name="slam2d_ros" type="slam2d_ros" output="screen" >
        <param name="scan_topic" value="$(arg scan_topic)"/>
        

        <param name="d_thresh"  value="$(arg d_thresh)"/>
        <param name="a_thresh"  value="$(arg a_thresh)"/>
        <param name="mrange"    value="$(arg mrange)"/>

        <param name="use_compression" value="$(arg use_compression)"/>

        <param name="create_summary"     value="true"/>
        <param name="map_publish_period" value="0.01" />
    </node>
</launch>`

and node graph same this https://github.com/iris-ua/iris_lama_ros/issues/40#issuecomment-861245936

jcmonteiro, 2021-06-15:
Is the `scan_topic` defined somewhere else?

IST-DEV20, 2021-06-15:
`  <?xml version="1.0"?>
<launch>

    <arg name="scan_topic" default="/scan"/>
    <arg name="rosbag" />
    <arg name="particles" default="30"/>
    <arg name="d_thresh"  default="0.5"/>
    <arg name="a_thresh"  default="0.5"/>
    <arg name="mrange"    default="20"/>

    <arg name="threads"   default="4"/>
    <arg name="use_compression" default="false"/>

    <param name="/use_sim_time" value="true"/>

    <node pkg="iris_lama_ros" name="slam2d_ros" type="slam2d_ros" output="screen" >
        <param name="scan_topic" value="$(arg scan_topic)"/>
        

        <param name="d_thresh"  value="$(arg d_thresh)"/>
        <param name="a_thresh"  value="$(arg a_thresh)"/>
        <param name="mrange"    value="$(arg mrange)"/>

        <param name="use_compression" value="$(arg use_compression)"/>

        <param name="create_summary"     value="true"/>
        <param name="map_publish_period" value="0.01" />
    </node>
</launch>
`

jcmonteiro, 2021-06-15:
And are you changing the topic by passing the argument to the launch? Are you really publishing to `/scan`?

IST-DEV20, 2021-06-16:
@jcmonteiro, Sorry for reply rate. Update I fixed problem by set parameter  `/use_sim_time` to `false` cause when I launch with  real hardware and set `/use_sim_time` to `true` no topic sim_time. 

Thank you for your help and support

closed 2021-06-16 by IST-DEV20

jcmonteiro, 2021-06-16:
Ooh, missed that one in your launch file. In practice, you should avoid `use_sim_time` in "inner" launch files and have it only in a top-level launch that calls for the others.