iris-ua/iris_lama_ros #41: Problems on integration of localization to ros navigation stack (move-base)

jojo89776, 2021-06-28:
First of all, I really appreciate the release of this wonderful mapping and localization package.
Currently, I'm working on an exchange amcl-based localization tool with this iris-lama-ros package.
However, when I was using this package with move base (ros navigation stack), I got the error message "Extrapolation Error: Lookup would require extrapolation into the future". (When using amcl-based localization, there was no error).
Do you have any idea to solve, or did you have any experience with such problems?
(I tried comment tf-broadcaster on localization node, but nothing changed).
It will be greatly helpful if you (or someone) can give any comments!

eupedrosa, 2021-06-28:
Hi @jojo89776, thanks for you kind words about LaMa.

I usually do not have an "Extrapolation Error" when using the localization node. Commenting the **tf-broadcaster** will not solve the problem, in fact, it makes it worse. Are you using melodic or noetic?

There is a parameter called ```~transform_tolerance```, try increase it to a higher value than ```0.1``` (eg 0.2 or 0.3).

jojo89776, 2021-06-29:
Thank you for the comment. 
Actually, when I used localization only, there was no error and the algorithm worked perfectly. 
This error occurs when I was using the lama-ros localization with move-base (Extrapolation error on local planner). 
I will change transform tolerance and figure out the problems.