iris-ua/iris_lama_ros #42: Large TF delay in odm to map transform

JordySehn, 2021-06-30:
First of all, great package!

So far (in simulation) I am getting very good localization results compared to amcl on our robot.

I did notice though that the transform delay from the odom to map transform seems unusually high to me? Using tf_monitor I get about 0.4 seconds average delay from odom -> map.

In contrast, amcl seems to only have a delay of about 0.1 seconds or so. Any idea why this would be? Have other people noticed similar transform delays?

JordySehn, 2021-06-30:
ahh you know what, the problem was just the transform tolerance parameter was set too low for me.

Bumping it up a tad brought the delay way down. Thanks!

closed 2021-06-30 by JordySehn

eupedrosa, 2021-06-30:
Glad I could help :smile:

Combinacijus, 2021-07-24:
TF delay could also happen if you built from source in with a Debug flag it's drastically slower than using [Release flag](https://answers.ros.org/question/311472/how-to-build-release-version-of-ros-node/)