iris-ua/iris_lama_ros #44: Trnasforming Maps Issue

yairBAR-caja, 2021-08-14:
Hi,

Thanks for this amazing pkg. It truly is remarkable.

I have an issue im struggling with and was wondering perhaps you might have some insight into. 
I'm not sure weather this is an actual issue with lama though and also i have a work around for it but was hoping to learn from this issue and maybe perhaps others who read this would too.

**The issue:**
I used slam2d_offline to generate a map. The generated map is great but the origin of the map is set at the initial pose of the robot where the recorded bag started. However, the rest of the robot works in a different global frame (called 'world') and so in-order for me to use lama's pose estimation i needed to setup a static transformer from lama's frame to 'world' frame. I was able to find the transformation from the map's origin to world's origin and used a static transformer to transform lama's pose to the world pose. In this setup everything works great.
  However, I wanted to avoid using a static transformer altogether by moving the map's origin (the one that was obtained through slam2d_offline) directly to my world origin. to do this, i simply set up a small python node and used tf2 to calculate the world origin w.r.t the pgm file's origin which is the left most bottom pixel. I then moved the origin of the map to the world's origin by resetting the "origin" field in the accompanying yaml to the pgm image to be that of the world frame.

When i do this, for some reason, lama's pose gets skewed over time (of course i also drop the code that transform lama's pose to world since now 'map' frame is the same 'world')

I'll note that in both cases, i use the '/initialpose' topic to force the starting position of the robot from a known position due to the fact that the robot does not always start at 0,0 in the map's frame. 

Maybe you have some insight as to why this is happening?   

Some technical data:

**Works great:**
slam map frame origin (as given by slam2d_offline):
origin: [-11.200000, -5.600000, 0.000000]
^ this origin is w.r.t. pgm left most bottom pixel

static transformation from slam map origin to world origin:
<node pkg="tf2_ros" type="static_transform_publisher" name="slam_map_to_world" 
        args="4.132492469511471 14.500942565021678 0.0 0.0 0.0 -0.7014344432926976 0.7127339768543826 /world /map" />

Then transform lama's pose to world. this works great. but feels like taking extra steps. 

**Gets skewed overtime:**
slam map frame origin after calculating world origin w.r.t pgm left most bottom pixel:
origin: [-1.645762293,25.6100276713,-1.55481620315]

Then i only use 'map' frame because there is no need for 'world' frame.

What could be causing lama's pose to skew?

tl;dr
Using a static transformer to transform lama's pose to a world pose works great.
However changing the map's origin to be that of world's origin and dropping the static transformer produces skewed results.
The origin transformation is 100% correct, what could be the issue?

Thanks for reading.

eupedrosa, 2021-08-14:
Hi,

Thanks for trying out LaMa.

I may not be able to fully understand the problem, but I have a few questions:
* Why does you robot need a specific origin world pose?
* How did you calculate the skew? Visual observation?

Furthermore, I believe the rotation componente of `origin: [-1.645762293,25.6100276713,-1.55481620315]`, which is the last one, does not actually rotate the map.

When creating the map you can always define an initial pose. You have the parameters `initial_pose_x`, `initial_pose_y` and `initial_pose_a` to define the initial pose. You may try using the negative of `origin` that you already calculated.

yairBAR-caja, 2021-08-14:
Hi eupedrosa, thanks for answering.

the world frame is shared between this robot and other robots like it.
the skew between the slam map and the world is achieved via exporting the pgm to CAD software and mapping out known and measured features in the scanned map (w.r.t. world origin). 
Im confident that the transform is correct because it is evident by the fact that a static transform is able to achieve good localization. 
i think the rotation component of origin rotates the origin inside the map. i have verified this in rviz, comparing images of the static transformer 'world' origin to the transformed map origin using the yaml. however i see what you are saying, in map_server they state:
"Many parts of the system currently ignore yaw"
Which might be the issue indeed maybe something is going wrong behind the scenes.

regarding initial pose during map creation, I didnt even notice its possible, kept walking through the code in loc2d instead. that's a very good idea ill try it and report back. Thanks for the tip.

yairBAR-caja, 2021-08-16:
@eupedrosa (Any relation to Dani Pedrosa btw? hehe) 
Worked like a charm. Thank you! 
Setting issue as closed, even though it was never an actual issue with lama.

closed 2021-08-16 by yairBAR-caja