iris-ua/iris_lama_ros #45: Message Filter Dropping all the messages

zainmehdi, 2021-08-25:
Hi I am trying to run Iris LAMA with YD Lidar X2L on a rapsberry pi. When I run online slam node I always get this message.

`MessageFilter [target=odom ]: Dropped 100.00% of messages so far. Please turn the [ros.iris_lama_ros.message_filter] rosconsole logger to DEBUG for more information.
`
Can there be any hints on whats wrong here ?

Thanks

eupedrosa, 2021-08-25:
Hi @zainmehdi.

That usually happens when the transformation frames are not correct. Please make sure you have the `base_frame_id` and `odom_frame_id` properly set, and also the `scan_topic` set to the correct laser topic.

By the way, which ROS version are you using?

zainmehdi, 2021-08-25:
hi @eupedrosa thanks for the reply. I am using ROS Noetic and Ubuntu Mate,

eupedrosa, 2021-08-25:
Remember to not use `/` in the name of the frames, for example: `base_frame_id=base_link` not `base_frame_id=/base_link`

zainmehdi, 2021-08-25:
Thanks alot. I will try and share my results.

zainmehdi, 2021-08-25:
I tried following command 
`rosrun iris_lama_ros slam2d_ros scan_topic:=scan base_frame_id:=base_footprint odom_frame_id:=odom`

YDLidar is publishing a static transform from base_footprint to laser_frame but still I get the same error. Is there anything wrong in the transforms ?

eupedrosa, 2021-08-25:
For the `scan_topic` try `scan_topic:=/scan`

zainmehdi, 2021-08-25:
Same result :(

eupedrosa, 2021-08-25:
Can you provide a small bag so I can try understand what it is happening?

zainmehdi, 2021-08-25:
Here is the bag file. 
https://drive.google.com/file/d/1udN77bSRptg_qSVWx9SdR7Fq7WeItFIV/view?usp=sharing
Thanks

eupedrosa, 2021-08-25:
@zainmehdi, you bag is missing the `/tf_static` 

Bu can you try with the following command:

`rosrun iris_lama_ros slam2d_ros _scan_topic:=/scan _base_frame_id:=base_footprint _odom_frame_id:=odom`

The parameters need the `_` prefix before their name otherwise they are interpreted as topic and not parameter.