iris-ua/iris_lama_ros #48: Suggestion: time based update

borsasd, 2021-11-17:
Hi!

First of all I would like to say that I'm very pleased with this library. We previously used gmapping on our mobile robot, but I found this package, tried it, and was blown away, because it seems to be better than gmapping in about every way. Also, loc2d works better than amcl for me as well. ROS integration is great too. So thanks for sharing this amazing software. I really think this should be the default mapping and localization in ROS.

On to my suggestion. I think it would be really useful to have time based updates for localization (maybe mapping too), kindof like gmapping's "temporalUpdate" parameter. So it would be possible to have a fixed frequency updating of the position even if the robot isn't moving based on the odometry. I know I could just call the "request_nomotion_update" service, but it would be nice to have a built-in option to do this periodically.

I think this won't be too difficult to do, so is there anything I'm missing here? Maybe there is some reason this is not already an option? Or do you think it wouldn't be useful?

Thanks

eupedrosa, 2021-11-17:
Hi @borsasd, thank you for your kind words.

A temporal update is not difficult to implement, so I will consider your suggesting.
I do not have a specific reason for never adding this feature and I'm aware that, for example, gmapping has it. 

Again, thank you for the suggestion. I will implement it once I have some spare time.

borsasd, 2021-11-18:
Thank you very much. I will study the code, and see if I can help with that.

eupedrosa, 2021-11-19:
@borsasd, I added the time base update to the localization node. Please test is and give me your feedback.

borsasd, 2021-11-19:
@eupedrosa, I tested it and works perfectly as far as I can tell. Does exactly what I wanted. Now the robot corrects its position and orientation after stopping with slipping wheels (it would frequently stay in the wrong place after that, as it did not update without wheel motion). Also, when started with the "do_global_loc" parameter, it can automatically locate itself on the map, without moving at all, I think this is very useful too (not perfect tho, sometimes can fail).

Also, I noticed that the README says that global localization is not implemented yet, maybe it should be removed, as it seems to be implemented indeed.

Thank you very much for your time on doing my request.

eupedrosa, 2021-11-19:
Great to know it works.

The global localization is not perfect, I know. Because it uses random particles there is always a change it will fail. To improve your changes you can always increase `gloc_particles`.