iris-ua/iris_lama_ros #49: Output odometry data in file format

zainmehdi, 2021-11-24:
Hi,
I was wondering if its possible to output odometry/graph data in the form of a file very much like g2o etc. I am planning to utilize offline map correction like in this package https://github.com/SMRT-AIST/interactive_slam . 

Looking forward to hearing from you

Thanks

eupedrosa, 2021-11-24:
Hi @zainmehdi.
> I was wondering if its possible to output odometry/graph data in the form of a file very much like g2o etc.

For the slam2d node?

Instead of having an internal method to save the poses and write them later, I could publish the pose of the robot and an external node can save them anyway it wants. This is something that I am more willing to implement (the pose publishing).
In fact, the pose publishing is on my TODO list.

zainmehdi, 2021-11-24:
Thanks alot @eupedrosa I think that would be really handy.

eupedrosa, 2021-11-24:
@zainmehdi, the pose is now publish after an update. I hope you find it helpful.

zainmehdi, 2021-11-29:
@eupedrosa thanks alot :)

closed 2021-11-29 by zainmehdi