iris-ua/iris_lama_ros #50: Galactic support

robvoi, 2022-03-19:
Is this node supported on galactic and the current nav2 version?

eupedrosa, 2022-03-21:
Hello @robvoi.

At the moment ROS2 is not supported. It will be in the future, but not right now. There is a branch for `dashing` and `eloquent` but they are not currently maintained.

LaMa is independent of ROS, so you can create your own port if you have the time to spare.

robvoi, 2022-03-21:
Thanks a lot.
Would really be curious too see it in action.

Time is one thing. Capability the other.  :-) 

> Hello @robvoi.
> 
> At the moment ROS2 is not supported. It will be in the future, but not right now. There is a branch for `dashing` and `eloquent` but they are not currently maintained.
> 
> LaMa is independent of ROS, so you can create your own port if you have the time to spare.

closed 2022-03-21 by robvoi