iris-ua/iris_lama_ros #52: Cant compile on Raspberry Pi - Noetic

JanezCim, 2022-05-16:
Hey @eupedrosa, i have a problem compiling iris on rpi4, heres what im doing:

Raspberry pi 4 rev 1.4

```
ubuntu@pi-focal:~/iris_slam_ws/src$ ls
iris_lama  iris_lama_ros
ubuntu@pi-focal:~/iris_slam_ws/src/iris_lama_ros$ git checkout noetic-devel
ubuntu@pi-focal:~/iris_slam_ws/src/iris_lama$ git status
On branch master
ubuntu@pi-focal:~/iris_slam_ws$ catkin config --extend /opt/ros/noetic
ubuntu@pi-focal:~/iris_slam_ws$ rosdep install --from-paths src --ignore-src -r -y
#All required rosdeps installed successfully
ubuntu@pi-focal:~/iris_slam_ws$ catkin build
...              
_____________________________________________________________________________________________________
Errors     << iris_lama:make /home/ubuntu/iris_ws/logs/iris_lama/build.make.000.log                  
/home/ubuntu/iris_ws/src/iris_lama/src/loc2d.cpp: In member function ‘void lama::Loc2D::addSamplingCovariance(const Ptr&)’:
/home/ubuntu/iris_ws/src/iris_lama/src/loc2d.cpp:211:55: error: no matching function for call to ‘max(unsigned int, long unsigned int)’
  211 |     const size_t step = std::max(num_points / 100, 1ul);

```

eupedrosa, 2022-05-16:
Hi @JanezCim.

It compiles fine on my normal computer. There must be some weird compiler behaviour.
Try changing to
```c++
const size_t step = std::max<size_t>(num_points / 100, 1ul);
```
or
```c++
const size_t step = std::max((size_t)(num_points / 100), (size_t)(1ul));
```

There may be some wrong type deduction from the compiler. Please let me know if it fixes anything.

eupedrosa, 2022-05-25:
That is an ABI warning, as I described [here](https://stackoverflow.com/questions/48149323/what-does-the-gcc-warning-project-parameter-passing-for-x-changed-in-gcc-7-1-m).

 Maybe recompile everthing?

JanezCim, 2022-05-26:
deleted and recompiled everything, the warning is still showing. I'd simply ignore it for now and call this solved

closed 2022-09-22 by eupedrosa