iris-ua/iris_lama_ros #55: new version of localization mode cost alot of time on startup

ZOUJIASHUAI, 2022-09-21:
hello @efernandez ,
does the new version of iris increased the localization perfomance?seems it cost a lot of time on startup if I use a 50x50m 0.02res map and on initpose.

efernandez, 2022-09-21:
@ZOUJIASHUAI I don't now. I don't own this repo and I don't use this software either. I only evaluated it a while ago.

Either way, if you want someone to look into that, you should provide a way to reproduce and share the performance results before the new version and after, so ppl can see the actual degradation in performance you've experienced.

eupedrosa, 2022-09-21:
Hi @ZOUJIASHUAI,

Can you be more specific with:
- the cpu that you are using
- the value of the parameter `l2_max`

A 50x50m with 0.02m of resolution can take a while depending on the value of `l2_max`.

eupedrosa, 2022-09-22:
Hi @ZOUJIASHUAI.

You should use a higher `l2_max`. I recommend the use of `0.5`.

Your build must be in debug mode. Go to your catkin workspace and run 
`catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release`.
Then recompile the code.

closed 2022-09-24 by ZOUJIASHUAI