iris-ua/iris_lama_ros #57: new lidar odom compiler error

ZOUJIASHUAI, 2024-04-25:
Errors     << iris_lama_ros:make /home/jason/loc_ws/logs/iris_lama_ros/build.make.005.log                                                                                                                                                                                        
In file included from /home/jason/loc_ws/devel/include/lama/lidar_odometry_2d.h:36:0,
                 from /home/jason/loc_ws/src/iris_lama_ros/include/lama/ros/lidar_odometry2d_ros.h:55,
                 from /home/jason/loc_ws/src/iris_lama_ros/src/lidar_odometry2d_ros.cpp:37:
/home/jason/loc_ws/devel/include/lama/types.h:101:27: error: redefinition of ‘template<class T> int lama::sign(T)’
 template <typename T> int sign(T val) {
                           ^~~~
In file included from /opt/ros/melodic/include/lama/image.h:38:0,
                 from /home/jason/loc_ws/src/iris_lama_ros/src/lidar_odometry2d_ros.cpp:34:
/opt/ros/melodic/include/lama/types.h:89:27: note: ‘template<class T> int lama::sign(T)’ previously declared here
 template <typename T> int sign(T val) {
                           ^~~~
In file included from /home/jason/loc_ws/devel/include/lama/lidar_odometry_2d.h:36:0,
                 from /home/jason/loc_ws/src/iris_lama_ros/include/lama/ros/lidar_odometry2d_ros.h:55,
                 from /home/jason/loc_ws/src/iris_lama_ros/src/lidar_odometry2d_ros.cpp:37:
/home/jason/loc_ws/devel/include/lama/types.h:105:8: error: redefinition of ‘struct lama::KeyHash’
 struct KeyHash {
        ^~~~~~~
In file included from /opt/ros/melodic/include/lama/image.h:38:0,
                 from /home/jason/loc_ws/src/iris_lama_ros/src/lidar_odometry2d_ros.cpp:34:
/opt/ros/melodic/include/lama/types.h:93:8: note: previous definition of ‘struct lama::KeyHash’
 struct KeyHash {
        ^~~~~~~
In file included from /home/jason/loc_ws/devel/include/lama/lidar_odometry_2d.h:36:0,
                 from /home/jason/loc_ws/src/iris_lama_ros/include/lama/ros/lidar_odometry2d_ros.h:55,
                 from /home/jason/loc_ws/src/iris_lama_ros/src/lidar_odometry2d_ros.cpp:37:
/home/jason/loc_ws/devel/include/lama/types.h:111:8: error: redefinition of ‘struct lama::PointCloudXYZ’
 struct PointCloudXYZ {
        ^~~~~~~~~~~~~
In file included from /opt/ros/melodic/include/lama/image.h:38:0,
                 from /home/jason/loc_ws/src/iris_lama_ros/src/lidar_odometry2d_ros.cpp:34:
/opt/ros/melodic/include/lama/types.h:99:8: note: previous definition of ‘struct lama::PointCloudXYZ’
 struct PointCloudXYZ {
        ^~~~~~~~~~~~~
In file included from /home/jason/loc_ws/devel/include/lama/lidar_odometry_2d.h:36:0,
                 from /home/jason/loc_ws/src/iris_lama_ros/include/lama/ros/lidar_odometry2d_ros.h:55,
                 from /home/jason/loc_ws/src/iris_lama_ros/src/lidar_odometry2d_ros.cpp:37:
/home/jason/loc_ws/devel/include/lama/types.h:122:8: error: redefinition of ‘struct lama::PolygonMesh’
 struct PolygonMesh {
        ^~~~~~~~~~~
In file included from /opt/ros/melodic/include/lama/image.h:38:0,
                 from /home/jason/loc_ws/src/iris_lama_ros/src/lidar_odometry2d_ros.cpp:34:
/opt/ros/melodic/include/lama/types.h:110:8: note: previous definition of ‘struct lama::PolygonMesh’
 struct PolygonMesh {
        ^~~~~~~~~~~
make[2]: *** [src/CMakeFiles/lidar_odometry2d_ros.dir/lidar_odometry2d_ros.cpp.o] Error 1
make[1]: *** [src/CMakeFiles/lidar_odometry2d_ros.dir/all] Error 2

ZOUJIASHUAI, 2024-04-25:
@eupedrosa please help

eupedrosa, 2024-04-25:
Are you also using the most recent version of [iris_lama](github.com/iris-ua/iris_lama) ?

ZOUJIASHUAI, 2024-04-30:
> Are you also using the most recent version of [iris_lama](github.com/iris-ua/iris_lama) ?

yes i did

ZOUJIASHUAI, 2024-05-05:
@eupedrosa can you compile iris lama & iris lama ros success?

eupedrosa, 2024-05-05:
Yes, it compiles. However, I only tested with ROS Noetic. Are you using a different distro?

ZOUJIASHUAI, 2024-05-08:
> Yes, it compiles. However, I only tested with ROS Noetic. Are you using a different distro?
@eupedrosa 
 I compiles with melodic，I will test it on Noetic

ZOUJIASHUAI, 2024-05-14:
hello @eupedrosa 
I compiled iris lama success on noetic,but failed on melodic ,can you try to make it work on melodic,then I can test it on factory

eupedrosa, 2024-05-14:
I did a fresh installation with melodic and it compiled. Are you using ubuntu 18.04?
I am not able to replicate your problem.